EmbeddedNim/picostdlib #6: example usb

Martinix75, 2021-08-31:
thanks Jeson for the very useful getCharWithTimeout () implementation !!
as a test I wrote a small test program, if you want to include it among the examples! 
[serialUsb.nim.zip](https://github.com/beef331/picostdlib/files/7083832/serialUsb.nim.zip)

beef331, 2021-09-08:
I'd like a PR if you have the time.

Martinix75, 2021-09-09:
PR? sorry but my English is not the best, what do you mean by PR?

beef331, 2021-09-09:
A pull request, submiting the addition to the repo with git.

Martinix75, 2021-09-09:
in fact it is one of the things that I have to learn well to do pull requests. I've never done them. but I promise that I study and at least I try (it's now past to learn)

Martinix75, 2021-09-13:
andrea@tec-martin:~/Documenti/Git/picostdlib/examples> git push -u origin master 
remote: Invalid username or password.

closed 2021-09-13 by Martinix75