EmbeddedNim/picostdlib #8: example usb2

Martinix75, 2021-09-13:
Hello! I tried to enter the sample code usb, but git forbids me:

andrea@tec-martin:~/Documenti/Git/picostdlib> git push --set-upstream origin Martinix75
remote: Invalid username or password.
fatal: Autenticazione non riuscita per 'https://github.com/beef331/picostdlib.git/'
andrea@tec-martin:~/Documenti/Git/picostdlib> 

I'm learning these days to use git, so chances are I'm still doing something wrong! :(

beef331, 2021-09-13:
I believe your `origin` is my repo, fork it then do `git remote add-url devel $yourGitUrlToFork`, the `git push --set-upstream devel $YourGitBranch`

Martinix75, 2021-09-14:
maybe I succeeded !!! let me know if the hello_ub file has arrived :))

beef331, 2021-09-14:
You did it! And also showed me that there needs to be sugar for the gpio as such i made https://github.com/beef331/picostdlib/blob/master/src/picostdlib/gpio.nim#L66
which is used like `setupGpio(led1, 25, Out)`