EmbeddedNim/picostdlib #17: multi-core rp2024

Martinix75, 2021-12-03:
Hi Jason,
I wanted to do some multicore tests (it's a beautiful thing about rp2024 !!). But as usual they are a bit complex for me to wrapped.
If you have time you could take a look?
The functions are:
• void multicore_reset_core1 (void)
• void multicore_launch_core1 (void (* entry) (void))
• void multicore_launch_core1_with_stack (void (* entry) (void), uint32_t * stack_bottom, size_t stack_size_bytes)
• void multicore_launch_core1_raw (void (* entry) (void), uint32_t * sp, uint32_t vector_table)

Pagi 193 of the manual.
thanks and bye  Martin A.

beef331, 2021-12-05:
I assume the main issue is just `void (* entry) (void)` which in Nim I believe is just `proc(){.cdecl.}`, all the other stuff should be fairly self explanatory `uint32_t*` -> `ptr uint32`, `size_t` -> `csize_t`, `uint32_t` -> `uint32`.

Martinix75, 2021-12-06:
Yes, exactly, that was my big problem (I had no idea how to do it!).
In the Christmas holidays I would try to do! Thank you
(and Merry Christmas in advance)

Martinix75, 2021-12-09:
Hi Jeson,
I had some time today and tried to do the job.
The rap of:
void multicore_launch_core1 (void (* entry) (void)) -> proc multicorelaunchcore1 * () {.cdecl, importc: "multicore_launch_core1".}

But the compiler says:
/home/andrea/progpiconim/hellomulticore/src/helloMulticore.nim(18, 21) Error: Type Mismatch: Got <proc () {. NosideEffect, GCSafe, Locks: 0.}>
But Expected One of:
Proc Multicorelaunchcore1 ()
  FIRST TYPE MISMATCH AT POSITION: 1
  Extra Argument Given

The program is taken from the one example in the manual:
import picostdlib/[gpio, multicore]
import picostdlib

const FlagValue: uint32 = 123

proc CoreOneActv() =
  multicoreFifoPushBlocking(FlagValue)
  var g: uint32 = multicoreFifoPopBlocking()
  if g != FlagValue:
    print("Male!! il Core 1 Non funziona  Bene!!" & '\n')
  else:
    print("Ottimo!! Il Core 1 Funziona!!" & '\n')

stdioInitAll()
print("Ciao Multicore!" & '\n')
multicoreLaunchCore1(CoreOneActv) (-->> LINE 18 <<---)
var g: uint32 = multicoreFifoPopBlocking()
if g != FlagValue:
  print("Male!! il Core 0 Non funziona  Bene!!" & '\n')
else:
  multicoreFifoPopBlocking(FlagValue)
  print("Ottimo!! Il Core 0 Funziona!!" & '\n')

I think it's not the right Wrapp as:
proc multicoreLaunchCore1*() {.cdecl, importC: "multicore_launch_core1".}

But it's too complicated for my knowledge 
bye bye by A.Martin

Martinix75, 2021-12-09:
I believe Dia Ver solved (I had never seen functions to functions !!): 
proc multicoreLaunchCore1*() {.cDecl, importC: "multicore_launch_core1".}
to
proc multicoreLaunchCore1*( function: pointer) {.cDecl, importC: "multicore_launch_core1".}

though I tell me if I did the right thing, because I'm not sure !! 
nim compile, but now i make some tests :)

Martinix75, 2021-12-10:
I ruined what you said (always in the hypothesis that has done well) 

{.push header: "pico/multicore.h".} 
type
  ThreadFunc* = proc() {.cDecl.}
proc multicoreLaunchCore1*(p: ThreadFunc) {.importC: "multicore_launch_core1".}
etc...

but the compiler tel me:
/home/andrea/ProgPicoNim/hellomulticore/src/hellomulticore.nim(27, 21) Error: type mismatch: got <proc (){.noSideEffect, gcsafe, locks: 0.}>
but expected one of:
proc multicoreLaunchCore1(p: ThreadFunc)
  first type mismatch at position: 1
  required type for p: ThreadFunc
  but expression 'CoreOneActv' is of type: proc (){.noSideEffect, gcsafe, locks: 0.}

expression: multicoreLaunchCore1(CoreOneActv)
oserr.nim(94)            raiseOSError
(## line 27 --> multicoreLaunchCore1(CoreOneActv) #attiva il core 1 ## CoreOneActv is a proc! ##)

For now it works (and it seems good !!) with: 
proc multicoreLaunchCore1*(function: pointer) {.cDecl, importC: "multicore_launch_core1".}

And it seems to work well on the two cores 
example output multi core..
[core0 120]␍␊
(core1 15)␍␊
[core0 122]␍␊
(core1 16)␍␊
(core1 17)␍␊
[core0 124]␍␊
(core1 18)␍␊
[core0 126]␍␊
(core1 19)␍␊
(core1 20)␍␊
[core0 128]␍␊
[core0 130]␍␊
[core0 132]␍␊

beef331, 2021-12-10:
`CoreOneActv` should be `proc CoreOneActv(){.cdecl.}` and it should work without the ugly `pointer`, this is how we deal with C interop.

Martinix75, 2021-12-10:
wow now run without "ugly" pointer
(## proc CoreOneActv() {.cdecl.} =
  var x = 0
  multicoreFifoPushBlocking(FlagValue).....##)

This was difficult for my level !! thank you!!

Martinix75, 2021-12-24:
Created wrapper for multicore and with a simple demonstration example, everything seems to work well!
Merry Christmas and Happy 2022

closed 2021-12-24 by beef331