EmbeddedNim/picostdlib #19: Add timeUs32() end timeUs64()

Martinix75, 2021-12-27:
Hi Jason I added two procedures (timeUs32() & 64) in the existing "time.nim" file creating a type structure:
{.push Header: "Something1".}
aaa
bbb
{.pop.}

{.push Header: "Something2".}
ccc
ddd
{.pop.}

But aren't they sure it's the correct thing to do, or if it is better to create a new Nim file? 
P.S. He says we can increase version from 0.2.6 -> 0.2.7?

beef331, 2021-12-27:
I do not know where those functions come from and no clue what they're for, have a link to them?

Martinix75, 2021-12-28:
Hi good morning.
The functions are in "src / rp2_common / hardware_timer / includes / hardware / timer.h" and are used to check how long the microntrolore is turned on (page 182 of the handle).
I have already done and tested the wrapper, only that github doesn't like it (I think).
I wrote the wrapper in "time.nim" and it works. 

....
{.push header:"pico/time.h".}
.....
  ## =========  ======
proc cancel*(alarm: AlarmId): bool {.importC: "cancel_alarm".}
  ## Cancel an alarm from the default alarm pool. 
  ## 
  ## **Parameters:**
  ## 
  ## =============  ====== 
  ## **alarm**       the alarm id 
  ## =============  ======
  ## 
  ## **Returns:**    true if the alarm was cancelled, false if it didn't exist 

{.pop.} ----> Here github from a mistake! 

{.push header: "hardware/timer.h".}
proc timeUs32*(): uint32 {.importC: "time_us_32".}
  ## Return time from the start of the microcontroller to microseconds (32-bit) 
  
proc timeUs64*(): uint64 {.importC: "time_us_64".}
  ## ## Return time from the start of the microcontroller to microseconds (64-bit)
{.pop.}

complete file:
[time.nim.txt](https://github.com/beef331/picostdlib/files/7783822/time.nim.txt)

I believe that because of the error Github has not acknowledged commit.

beef331, 2021-12-28:
It seems fine to be placed in `time.nim` so I await the PR.

Martinix75, 2021-12-28:
Check if it's okay! It should be all affixed now!

Martinix75, 2021-12-30:
I postponed the file TIME.NIM and the example file, I hope (2 files) and should work well.

Martinix75, 2021-12-30:
add only 1 commit (time. nim)

Martinix75, 2021-12-30:
and 2 commit for helotimestart.nim (2 separate commit)

P.S. A simple generator of random numbrs written in NIM (and based on UTIME32 ()) is coming only a few tests are missing but seems to work well

closed 2021-12-31 by beef331