EmbeddedNim/picostdlib #49: Example for Pico W

inviridi, 2022-08-27:
Hi guys, 

thanks for this lovely project; I really enjoyed the clean and useful examples! Would you mind showing how to use Nim with the Pico W, i.e. leveragint its wireless capabilities? My internet searches for this were fruitless so far and I think it would make a good addition to the repo.

Thanks a lot in advance!

Cheers,
Frank

beef331, 2022-08-27:
I do not have a PicoW, so I can only tell you what's required. You will need to wrap the C procedures. Examples of the C api are [here](https://github.com/raspberrypi/pico-examples#pico-w-networking) after wrapping the procedures and types that it uses you'll be capable of re implementing those  examples and sleep happily.

inviridi, 2022-08-28:
Thanks for your swift reply, @beef331! I'll try wrapping the C procedures then and sleep happily.

closed 2022-08-28 by inviridi