EmbeddedNim/picostdlib #55: Bunch of gcc warnings

auxym, 2022-10-03:
Nothing is broken, just an annoyance. Don't remember seeing this before, but currently my pico projects (even a fresh blinky created with piconim init) results in whole bunch of complaining from gcc (see below).

Not sure what changed. I tried older Nim versions and still getting the same. Maybe a new gcc version? I am running `arm-none-eabi-gcc (Arch Repository) 12.2.0`.

"Regular" nim compiles don't do this, I guess the pico CMake/make stuff might be passing different `-W` flags to gcc?

```
In file included from /home/francis/temp/proj/csource/build/nimcache/stdlib_system.nim.c:4:
/home/francis/temp/proj/csource/build/nimcache/stdlib_system.nim.c: In function 'align_system_1616':
/home/francis/temp/proj/csource/build/nimcache/stdlib_system.nim.c:281:52: warning: passing argument 3 of '__builtin_ssub_overflow' from incompatible pointer type [-Wincompatible-pointer-types]
  281 |                 if (nimSubInt(alignment, ((NI) 1), &TM__Q5wkpxktOdTGvlSRo9bzt9aw_3)) { raiseOverflow(); goto BeforeRet_;
      |                                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                    |
      |                                                    NI * {aka long int *}
/home/francis/.choosenim/toolchains/nim-1.6.0/lib/nimbase.h:589:64: note: in definition of macro 'nimSubInt'
  589 |     #define nimSubInt(a, b, res) __builtin_ssub_overflow(a, b, res)
      |                                                                ^~~
/home/francis/temp/proj/csource/build/nimcache/stdlib_system.nim.c:281:52: note: expected 'int *' but argument is of type 'NI *' {aka 'long int *'}
  281 |                 if (nimSubInt(alignment, ((NI) 1), &TM__Q5wkpxktOdTGvlSRo9bzt9aw_3)) { raiseOverflow(); goto BeforeRet_;
      |                                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/francis/.choosenim/toolchains/nim-1.6.0/lib/nimbase.h:589:64: note: in definition of macro 'nimSubInt'
  589 |     #define nimSubInt(a, b, res) __builtin_ssub_overflow(a, b, res)
      |                                                                ^~~
/home/francis/temp/proj/csource/build/nimcache/stdlib_system.nim.c:283:78: warning: passing argument 3 of '__builtin_sadd_overflow' from incompatible pointer type [-Wincompatible-pointer-types]
  283 |                 if (nimAddInt(address, (NI)(TM__Q5wkpxktOdTGvlSRo9bzt9aw_3), &TM__Q5wkpxktOdTGvlSRo9bzt9aw_4)) { raiseOverflow(); goto BeforeRet_;
      |                                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                              |
      |                                                                              NI * {aka long int *}
/home/francis/.choosenim/toolchains/nim-1.6.0/lib/nimbase.h:588:64: note: in definition of macro 'nimAddInt'
  588 |     #define nimAddInt(a, b, res) __builtin_sadd_overflow(a, b, res)
      |                                                                ^~~
/home/francis/temp/proj/csource/build/nimcache/stdlib_system.nim.c:283:78: note: expected 'int *' but argument is of type 'NI *' {aka 'long int *'}
  283 |                 if (nimAddInt(address, (NI)(TM__Q5wkpxktOdTGvlSRo9bzt9aw_3), &TM__Q5wkpxktOdTGvlSRo9bzt9aw_4)) { raiseOverflow(); goto BeforeRet_;
      |                                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/francis/.choosenim/toolchains/nim-1.6.0/lib/nimbase.h:588:64: note: in definition of macro 'nimAddInt'
  588 |     #define nimAddInt(a, b, res) __builtin_sadd_overflow(a, b, res)
      |                                                                ^~~
/home/francis/temp/proj/csource/build/nimcache/stdlib_system.nim.c:285:52: warning: passing argument 3 of '__builtin_ssub_overflow' from incompatible pointer type [-Wincompatible-pointer-types]
  285 |                 if (nimSubInt(alignment, ((NI) 1), &TM__Q5wkpxktOdTGvlSRo9bzt9aw_5)) { raiseOverflow(); goto BeforeRet_;
      |                                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                    |
      |                                                    NI * {aka long int *}
/home/francis/.choosenim/toolchains/nim-1.6.0/lib/nimbase.h:589:64: note: in definition of macro 'nimSubInt'
  589 |     #define nimSubInt(a, b, res) __builtin_ssub_overflow(a, b, res)
      |                                                                ^~~
/home/francis/temp/proj/csource/build/nimcache/stdlib_system.nim.c:285:52: note: expected 'int *' but argument is of type 'NI *' {aka 'long int *'}
  285 |                 if (nimSubInt(alignment, ((NI) 1), &TM__Q5wkpxktOdTGvlSRo9bzt9aw_5)) { raiseOverflow(); goto BeforeRet_;
      |                                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/francis/.choosenim/toolchains/nim-1.6.0/lib/nimbase.h:589:64: note: in definition of macro 'nimSubInt'
  589 |     #define nimSubInt(a, b, res) __builtin_ssub_overflow(a, b, res)
      |                                                                ^~~
/home/francis/temp/proj/csource/build/nimcache/stdlib_system.nim.c: In function 'nimDecRefIsLast':
/home/francis/temp/proj/csource/build/nimcache/stdlib_system.nim.c:323:61: warning: passing argument 3 of '__builtin_ssub_overflow' from incompatible pointer type [-Wincompatible-pointer-types]
  323 |                         if (nimSubInt((*cell).rc, ((NI) 8), &TM__Q5wkpxktOdTGvlSRo9bzt9aw_10)) { raiseOverflow(); goto BeforeRet_;
      |                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                             |
      |                                                             NI * {aka long int *}
/home/francis/.choosenim/toolchains/nim-1.6.0/lib/nimbase.h:589:64: note: in definition of macro 'nimSubInt'
  589 |     #define nimSubInt(a, b, res) __builtin_ssub_overflow(a, b, res)
      |                                                                ^~~
/home/francis/temp/proj/csource/build/nimcache/stdlib_system.nim.c:323:61: note: expected 'int *' but argument is of type 'NI *' {aka 'long int *'}
  323 |                         if (nimSubInt((*cell).rc, ((NI) 8), &TM__Q5wkpxktOdTGvlSRo9bzt9aw_10)) { raiseOverflow(); goto BeforeRet_;
      |                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

(hundreds more lines of similar warnings)
```

auxym, 2022-10-05:
Well this is fun. Adding `--cpu:arm` to the `nim c -c ...` command line in piconim removes the warning. It seems like Nim is creating bad codegen (for some other CPU target) without the `--cpu` CLI flag, even though the `cpu` option from config.nims appears to be correctly considered. Adding:

```nim
static:
  echo hostCPU
```
To the project file getting compiled results in `arm` being echoed in both cases, cpu CLI flag or not. Weird.

Seems like a compiler bug. I'll investigate a bit more and try to produce an MRE to open an issue in the compiler.

beef331, 2022-10-05:
Some flags dont behave properly from inside configs, so that could be the culprit.

auxym, 2022-10-05:
Yeah, looks like that issue. Are you aware of an existing issue in the compiler repo?

Workaround would probably be to "hardcode" the cpu flag in piconim. Weird because it's not happening with my samd21 project, using similar flags and same gcc. I'll look into it.

auxym, 2022-10-06:
So the only reason I wasn't seeing it happening in my other arm project is because I let nim do the full build, and it calls gcc with `-w` which hides all warnings (yay).

I have a MRE here: https://github.com/auxym/nim_incompat_pointer_bug

Will open a proper bug report in the nim compiler repo when I get a chance. I can confirm that the codegen is different when `--cpu:arm` is passed to Nim on the CLI vs in the config file, even though nim doesn't fully ignore the config (`static: echo hostCPU` still print `arm` during the build in both cases).

PR #56 (2022-10-08): Suppress gcc warnings when compiling nim sources
Resolves #55

1. With Nim 1.6.8, the warnings appear no matter whether `--cpu:arm` is passed  on the command line or not. I assume this is a fix to harmonize behavior between config and command line flags.

2. I asked Araq on Discord, and the incompatible pointer type warnings are expected and not an issue.

3. Nim itself, when it invokes gcc, always passes the `-w` flag which suppresses all warnings.

Therefore, I propose to imitate the Nim compiler behavior and pass `-w` to gcc for all Nim-generated C files. This is implemented via a `set_sour [... cut]
closed 2022-10-08 by beef331