EmbeddedNim/picostdlib #64: Added new bookstores how to do it?

Martinix75, 2023-07-05:
Hi guys with SDK 1.5.x added random numbers, it is true that I had written un my small library for them in the meantime (https://github.com/martiniix75/raspberry_pico/tree/main/libs). But since now there is the official one, I wrappata, added in the libreire and also added to "Piconim.nim"

pico_rand.nim
```nim
type
  rng_128_t* {.bycopy.} = object
    r*: array[2, uint64]
    
{.push header: "pico/rand.h".}  

proc getRand128*(rand128: ptr rng_128_t) {.importC: "get_rand_128".}
proc getRand64*(): uint64 {.importC: "get_rand_64".}
proc getBand32*(): uint32 {.importC: "get_rand_32".}
{.pop.}
```
end piconim.nim
```nim
type
  LinkableLib = enum
    stdio = "pico_stdlib"
    multicore = "pico_multicore"
    gpio = "pico_stdlib"
    adc = "hardware_adc"
    pio = "hardware_pio"
    dma = "hardware_dma"
    i2c = "hardware_i2c"
    rtc = "hardware_rtc"
    uart = "hardware_uart"
    spi = "hardware_spi"
    clock = "hardware_clocks"
    reset = "hardware_resets"
    flash = "hardware_flash"
    pwm = "hardware_pwm"
    interp = "hardware_interp"
    rand = "pico_rand" # <----- add this!!!!!
```
But the following mistake returns to me, or I make a very stupid error, or I have to change more (to test). So what should I change again ??? Thank you

'''
[ 12%] Built target ELF2UF2Build
[ 13%] Building C object CMakeFiles/cazzo.dir/nimcache/@m..@s..@s..@s.nimble@spkgs@spicostdlib-0.3.3@spicostdlib@sstdio.nim.c.obj
[ 14%] Building C object CMakeFiles/cazzo.dir/nimcache/@m..@s..@s..@sbin@snim@slib@sstd@sprivate@sdigitsutils.nim.c.obj
[ 15%] Building C object CMakeFiles/cazzo.dir/nimcache/@m..@s..@s..@sbin@snim@slib@ssystem.nim.c.obj
[ 17%] Building C object CMakeFiles/cazzo.dir/nimcache/@m..@s..@s..@sbin@snim@slib@ssystem@sdollars.nim.c.obj
[ 18%] Building C object CMakeFiles/cazzo.dir/nimcache/caz.c.obj
/home/andrea/ProgPicoNim/caz/csource/build/nimcache/caz.c:6:10: fatal error: pico/rand.h: No such file or directory
    6 | #include "pico/rand.h"
      |          ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/cazzo.dir/build.make:132: CMakeFiles/cazzo.dir/nimcache/caz.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:1509: CMakeFiles/caz.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
make: Leaving directory '/home/andrea/ProgPicoNim/caz/csource/build'
```
bye by Andrea

beef331, 2023-07-05:
https://github.com/EmbeddedNim/picostdlib/pull/62 will close this, we're just doing some final house cleaning on it.

Martinix75, 2023-07-06:
ok !! Maybe I will reopen it later, it's not urgent, for now I use my library!

closed 2023-07-06 by Martinix75