EmbeddedNim/picostdlib #68: Is SPI ported?

crystalthoughts, 2024-12-09:
Hello, is SPI part of the GPIO library ( I see a function mode referring to it) or is this not yet ported?
Thanks for the library!

crystalthoughts, 2024-12-10:
Yess I think that's a key piece needed to get most projects with a screen going :)
Have you tried using Futhark to speed up the porting process?
I'm not really experienced enough in embedded to do a good job contributing to this unfortunately, otherwise I would try!

crystalthoughts, 2024-12-10:
https://github.com/EmbeddedNim/picostdlib/pull/62

It looks like this pull request has it covered (seemed also to use Futhark)

crystalthoughts, 2024-12-12:
@daniel-j there are some downstream dependencies that aren't handled by the nimble file (pico-vfs -> littlefs)

crystalthoughts, 2024-12-12:
[image]
I also get this when I try the examples

daniel-j, 2024-12-16:
@Martinix75 Hello! Can you describe or show the error that occur?

@crystalthoughts Hmm, not sure what may be the issue. What are the error messages from the compilation or configuration? The screenshot only shows cmake failed but it doesn't show what cmake error it was. LittleFS is now loaded from within pico-vfs. Make sure to clone picostdlib recursively!

daniel-j, 2024-12-19:
> @daniel-j <https://github.com/daniel-j>
> Hi Daneil, sorry for the delay for the answer, but I was not at work!
> I am attaching you below the error that brings me back (everything).
> But I would say that the certain is "Fatal: 'sys/time.h' file not found ".
> I use OpenSUSE SLOWROLL/TW
> GCC-13 (GCC-13 Cross Arm None)
> Clang 19.x
> Nim 2.2.0
> Maybe only a few files/package is missing? I would not know.
>
>   Success:  futhark installed successfully.
>    Building picostdlib/piconim using c backend
> Setting up build/futharkgen

Hi

Make sure package cross-arm-none-newlib-devel is installed as that contains
sys/time.h for embedded.

Daniel

Martinix75, 2024-12-20:
Yes! The "cross-arm-none-newlib-devel (version 4.4.0.xxx)" pachet is installed.
And the file is in "/usr/arm-none-eobi/includa/sys/time.h"

daniel-j, 2024-12-26:
Sorry for the delay. What does `arm-none-eabi-gcc -print-sysroot` print? I pushed an update now to use `/usr/arm-none-eabi/include` as fallback if it exists.

Martinix75, 2025-01-07:
Hi @Daniel, sorry for the delay of the answer (and the test) but the Christmas holidays are a nightmare.
This morning I copied your library and after the Commit it works, or at least it is completed without errors! Thank you.
In these days I will try to make a plan and see if everything works. But for now thanks !!!

Martinix75, 2025-01-15:
Hi @daniel-j 
In these days I am a little playing with "spi.nim"
I noticed two things:
1 - The names are not consistent, while for i2c the names of the functions begin for i2c (eg i2csetu, i2cwritebloking ...) while for spi no (would be perhaps better if you start for spi?)
2- The function (SPI) write16Blocking does not work well (and on oscilloscope) by very strange results that are not read well from the SPI device that I have (MCP42050). I have allowed myself to write a nim procedure that seems to work well (for the writing of 16 continuous bit) but needs a particular setup (procedure) that I have written anyway.
When I try it well if you want I send it to you.

Hello from martinix